pavel-demin/stemlab-sdr-notes #2: Trying to get FT8 receiver working with @azwirko's perl scripts

bjornekelund, 2019-03-03:
I have access to a couple of STEMlab SDR boards and I am trying to get your software up and running on them. The HPSDR compatible multi band receiver works like a charm, with noticeable better performance than with the 125-14 but I am having trouble with the FT8 receiver/decoder. It runs and decodes plenty of signals (which I can see in the DECODES_date_time.txt files in /dev/shm) but there is no upload to PSKreporter and Andy's perl scripts also report nothing. Looking closer at the issue, I find that the files decode-ft8.log and upload-ft8.log only contains the "Recording...", "Sleeping..." etc. lines and no decodes. Is there a parameter setting I have missed or something?

pavel-demin, 2019-03-03:
The [PSK Reporter Statistics](https://www.pskreporter.info/cgi-bin/pskstats.pl) web page shows your callsign associated with the new STEMlab SDR FT8 TRX software. So, there was at least one upload that reached the PSK Reporter server.

Andy's Perl scripts were designed to run under Ubuntu. I don't know if they work with my SD card image based on Alpine Linux.

I'm currently running the FT8 application using the 20190224 SD card image. The preinstalled scripts and programs work without problem. The uploads happen every 5-6 minutes with a random time offset.

To enable uploads, the CALL and GRID variables should be specified in [upload-ft8.sh](https://github.com/pavel-demin/stemlab-sdr-notes/tree/master/projects/sdr_transceiver_ft8/app/upload-ft8.sh#L4-L5).

pavel-demin, 2019-03-03:
Here is how upload-ft8.log looks like when the uploads are enabled:
```
Sun Mar  3 16:00:01 UTC 2019
Sleeping ...
Sun Mar  3 16:00:28 UTC 2019
Processing ...
Uploading ...
Sun Mar  3 16:06:01 UTC 2019
Sleeping ...
Sun Mar  3 16:06:07 UTC 2019
Processing ...
Uploading ...
```

bjornekelund, 2019-03-03:
I see. I can not get the RP Ubuntu to boot on the SDR board. Not even using the most recent beta image. I myself could find my callsign (SM7IUN) on the PSK Reporter page you refer to. If there are any spots with that call, they are probably from earlier in the week. I use WSJT-X almost daily, and that reports all decodes. Regardless, I found the problem. I created the SD card on my Windows PC and modified the upload-ft8.sh while the card was in the PC's card reader. It seems these changes were completely ignored. I now changed the file with nano on the RP and now I can see "Uploading..." in upload-ft8.log! 

I have installed your image per your instructions and also modified the upload-ft8.sh but I did it when the SD card was connected to my PC and I now see that these changes have been ignored. Editing the file using nano on the RP made the changes "take" and it now works. It seems the Alpine system is completely reset every time I modify any of the files

bjornekelund, 2019-03-03:
I have another question - the reference oscillator on the RP running the FT8 decoder is 6.8ppm too high, resulting in all reported frequencies being 6.8ppm too low. Should I set "corr" in write-c2-files.cfg to plus or minus 6.8?

pavel-demin, 2019-03-03:
>I created the SD card on my Windows PC and modified the upload-ft8.sh while the card was in the PC's card reader. It seems these changes were completely ignored.

Looks like the editor that you used to edit upload-ft8.sh replaced the Unix end-of-line characters with the Windows end-of-line characters. So, an editor that doesn't change the end-of-line characters should be used for editing. For example, Notepad++.

> Should I set "corr" in write-c2-files.cfg to plus or minus 6.8?

I think that it should be minus 6.8.

bjornekelund, 2019-03-03:
Thanks. Makes sense. The correct calibration factor in PowerSDR is 0.99999322.

bjornekelund, 2019-03-03:
Is it normal that /etc/motd is reset at each reboot?

pavel-demin, 2019-03-03:
> Is it normal that /etc/motd is reset at each reboot?

I don't know. Have you tried to save your changes using `lbu commit -d`?

bjornekelund, 2019-03-03:
No. I am slowly rediscovering Linux after having been away from hands-on software for a very long time. Before last year, the last Unix machine I wrote code on was a VAX 11/782 running BSD 4.2. `lbu commit -d` worked perfectly. Thanks.

bjornekelund, 2019-03-04:
Update: Andy's scripts does not work with 1) an unmodified version of your software and 2) the latest version of RBN Aggregator. In spite of not really knowing perl I was able to correct the issues I now have a script that works and uploads all FT8 spots to RBN in addition to pskreporter. It however requires adding an extra print out (`echo "Done decoding..."` ) in decode-ft8.sh after the `wait`. I am happy to share with whomever is interested.

bjornekelund, 2019-03-05:
Update: Still under Alpine, I was able to design a perl script that broadcasts UDP packets using the WSJT-X datagram format. RBN Aggregator then picks them up and forwards them to the RBN network. However, to get a reliable and relevant trigger for the script I needed to add `echo "Done decoding"` after the `wait` in decode-ft8.sh. Since this is helpful for any script using `tail -f decode-ft8.log` to piggy-back on your decoder, would it be possible to add this to the master branch? Or do you prefer me to make a pull request?

pavel-demin, 2019-03-05:
> Since this is helpful for any script using `tail -f decode-ft8.log` to piggy-back on your decoder

I don't think that it's a good idea to keep a long running script connected via a pipe to `tail -f decode-ft8.log`. The process should somehow be restarted every time the log file is removed.

Could you try a solution that runs a cron job and looks for files created more than 1 minute ago. For example, here how it's done in [upload-ft8.sh](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_transceiver_ft8/app/upload-ft8.sh#L28-L36):
```
REPORT=report_$TIMESTAMP.txt

for file in `find . -name decodes_\*.txt -mmin +1`
do
  cat $file >> $REPORT
  rm -f $file
done

test -f $REPORT || exit

# upload report

rm -f $REPORT
```

I think that the following Perl code should find the files created more than 1 minute ago:
```
my $time = time - 60;
my $files = File::Find::Rule->file()->mtime(">$time")->in(".");
```

bjornekelund, 2019-03-05:
Fully agreed a much cleaner approach. I will look into this.

bjornekelund, 2019-03-08:
Problem solved. Actually far simpler than I thought. Just after the `wait`in `decode-ft8.sh` I call a perl script that parses the decode result file having the name 

`($x, $mn, $hr, $dy, $mo, $yr, $x, $x, $x) = gmtime(time - 60);`\
`$filename = "/dev/shm/" . sprintf("decodes_%02d%02d%02d_%02d%02d.txt", $yr - 100, $mo + 1, $dy, $hr, $mn);`

for decodes, converts them into UDP packets in WSJT-X format and sends them to the IP address and port of the computer running RBN Aggregator. Executing the script takes just a fraction of a second. 

I also modified `decode-ft8.sh` to copy three different recorder configurations to `/dev/shm`: `write-c2-files-day.cfg`, `write-c2-files-night.cfg`, and `write-c2-files-gray.cfg`, and then use my `radioday`utility to select the correct one. 

```$RECORDER write-c2-files-`$DIR/radioday $GRID $GRAYDURATION`.cfg``` 

Since it's a collection of files rather than a consolidated design I will probably publish it on my web site rather than GitHub.

bjornekelund, 2019-03-09:
One more question on decodes: With the 2.0 protocol Joe Taylor introduced the concept of hashed calls meaning that some long/irregular calls are only transmitted periodically and then replaced by a hash most of the time. In the WSJT-X UI hashed calls are displayed as "<AB1CDEF>". Since the current version of WSJT-X export the brackets as part of the call sign in its UDP broadcast, this cause problems in some other applications. In the decode stream from your FT8 receiver I see no such brackets. Are hashed calls ignored or properly taken care of (but forwarded without brackets) by your FT8 decoder?

pavel-demin, 2019-03-09:
After the following two commits the empty and <...> callsigns are filtered out:

https://github.com/pavel-demin/ft8d/commit/53f062c9f923bc5419c87d7023100fe34fbcf5af
https://github.com/pavel-demin/ft8d/commit/6117fb237f6f28cec397b765f2884e17785ab35a

I don't know whether the empty and <...> callsigns correspond to the hashed callsigns or not.

I'd suggest to have a look at the comments in the other issue about the FT8 application starting from [this comment](https://github.com/pavel-demin/red-pitaya-notes/issues/689#issuecomment-455925343). You'll find there some examples of empty and <...> callsigns.

bjornekelund, 2019-03-09:
Thanks. I don't know the reason for completely empty callsigns but in the WSJT-X UI, "<...>" instead of a proper callsign means the hash from a hashed callsign was received but the decoder was unable to convert it to an actual call sign since it has not yet received it in full. This may not be a problem since irregular callsigns (requiring hashing for message exchange) is periodically transmitted in full anyway. So if the STEMLab FT8 decoder does not include the mechanic to convert callsign hashes into the actual call sign, that just means fewer reports. It does not mean zero reports.

closed 2019-04-02 by pavel-demin