pavel-demin/stemlab-sdr-notes #3: Cyclic replacement of write-c2-files.cfg for FT8 receiver

bjornekelund, 2019-03-05:
One awkward limitation with the Red Pitaya is the fact that it supports eleven relevant radio frequency bands but only eight can be received at any one time. 

Since propagation on HF is largely dependent on the ionosphere it is however not relevant to listen to all eleven bands at all times. 

At night time propagation above 14MHz is limited, at day the same is true for frequencies below 7MHz. 

Myself I am using a four-stroke daily cycle (dusk, night, dawn, day) for my Red Pitaya Hermes receiver feeding my CW Skimmer Server. This makes it possible to cover all eleven bands at relevant times, still under the eight-band limitation.  

In my particular case dusk starts one hour before sunset, night starts one hour after sunset, dawn begins one hour before sunrise and day starts one hour after sunrise. By using sunset/sunrise, seasonal variations are automatically accounted for. 

A lot of skimmers around the globe are using this type of scheduling since it is a feature built into RBN Aggregator which controls the Hermes receiver via CW Skimmer Server.

Is this something you could consider adding to the FT8 receiver? Or would you suggest I just rewrite `decode-ft8.sh` to copy different `write-c2-files.cfg` to `/dev/shm` at relevant times?

| Band | 1.8 | 3.5 | 5.3 | 7 | 10 | 14 | 18 | 21 | 24 | 28 | 50 |
| --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- |
| Night | x | x | x | x | x | x | x | x |  |  |  | 
| Dusk/Dawn |  | x | x | x | x | x | x | x | x |  |  | 
| Day |  |  |  | x | x | x | x | x | x | x | x |

pavel-demin, 2019-03-05:
> Is this something you could consider adding to the FT8 receiver?

At the moment, I'm not planning to work on this functionality.

bjornekelund, 2019-03-05:
Understood. I will proceed with a fork.

bjornekelund, 2019-03-07:
Problem solved https://github.com/bjornekelund/radioday

closed 2019-04-02 by pavel-demin