pavel-demin/stemlab-sdr-notes #4: Possible defect in upload-to-pskreporter.c

bjornekelund, 2019-03-10:
Since I'm using your code as the base for my own upload-to-rbn utility, I think I have stumbled on a defect.

I believe the section:

``` 
rc = read_time(&src, &tm)
        && read_dbl(&src, &sync)
        && read_int(&src, &snr)
        && read_dbl(&src, &dt)
        && read_int(&src, &freq)
        && sscanf(src, "%6s %4s", call, grid); 
```
should read
```   
rc = read_time(&src, &tm)
        && read_dbl(&src, &sync)
        && read_int(&src, &snr)
        && read_dbl(&src, &dt)
        && read_int(&src, &freq)
        && sscanf(src, "%8s %4s", call, grid); 
```
`call` is `char[8]` and I noticed that long call signs occasionally got truncated, spilling over into `grid`.

pavel-demin, 2019-03-10:
Thanks for finding this problem. I forgot to port [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/8c0e8ce455ecc02e0181eec23b1a322c1f3297b6) from red-pitaya-notes to stemlab-sdr-notes.

pavel-demin, 2019-03-10:
It's fixed now with [this commit](https://github.com/pavel-demin/stemlab-sdr-notes/commit/50ad48d10092d22bcad4096f2adbe346886b9c20).

bjornekelund, 2019-03-12:
I believe the SD card image file referred to in https://pavel-demin.github.io/stemlab-sdr-notes/sdr-transceiver-ft8 is not updated to this new commit?

pavel-demin, 2019-04-02:
There is now a release with the updated upload-to-pskreporter.c:
https://github.com/pavel-demin/stemlab-sdr-notes/releases/tag/20190402

closed 2019-04-02 by pavel-demin