pavel-demin/stemlab-sdr-notes #5: Solution for uploading FT8 decodes to RBN

bjornekelund, 2019-03-11:
I have written an app [`upload-to-rbn`](https://github.com/bjornekelund/upload-to-rbn) that communicates with [RBN Aggregator](http://www.reversebeacon.net/pages/Aggregator+34) using the [WSJT-X UDP broadcast protocol](https://sourceforge.net/p/wsjt/wsjtx/ci/master/tree/NetworkMessage.hpp) to upload decodes from the FT8 receiver to the [Reverse Beacon Network](http://www.reversebeacon.net/index.php). To use it a modified [`decode-ft8.sh`](https://github.com/bjornekelund/upload-ft8-to-rbn/blob/master/decode-ft8.sh) is required. The modified `decode-ft8.sh` also makes use of a config file cycling utility [`radioday`](https://github.com/bjornekelund/radioday) to enable the use of different sets of frequency bands depending on sunlight status. The source code for upload-to-pskreporter and other parts of your software were great helps. Thank you.

bjornekelund, 2019-04-15:
Update: Since Pavel updated the receiver to cover up to 16 segments, there is no real need for the `radioday` utility anymore. The step-by-step description on my web site has been update to reflect this.

closed 2019-05-28 by pavel-demin

ledodd, 2019-06-05:
I can't seem to get my Community Login Activated so will try this email 
method.  Got the new 16 bit RedPitaya and got it working as an HF SDR 
but I want to use it as a spectrum Analyzer for monitoring 15 to 30 MHz 
waterfall for astronomy work.  Can't seem to find the software to do 
that. Any help appreciated.  Thanks
Larry K4LED