pavel-demin/stemlab-sdr-notes #6: Defect in upload-to-pskreporter

bjornekelund, 2019-05-15:
There is a defect in https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_transceiver_ft8/app/upload-to-pskreporter.c that manifests itself by re-using the previous grid in reports lacking one. The code segment in question is

    rc = read_time(&src, &tm)
        && read_dbl(&src, &sync)
        && read_int(&src, &snr)
        && read_dbl(&src, &dt)
        && read_int(&src, &freq)
        && sscanf(src, "%13s %4s", call, grid);

I solved it by adding `sprintf(grid, "    ");` just before. There may be more elegant ways.

pavel-demin, 2019-05-15:
Normally, the two lines just before the code that you've copied set the `call` and `grid` variables to empty strings. Here are these two lines:
```
call[0] = 0;
grid[0] = 0;
```

I've just tested the code with a file containing a few decodes and it works without any problem. If previous grid locator isn't reused in reports lacking one.

Could you please provide more information that would help me to reproduce this problem?

bjornekelund, 2019-05-15:
You are perfectly correct. I copied parts of your code for my upload-to-rbn utility but accidentally omitted those two lines and then wrongly assumed your code had the same issue. There is nothing wrong with your code. My apologies.

pavel-demin, 2019-05-15:
Thanks for the clarification.

closed 2019-05-15 by pavel-demin