pavel-demin/stemlab-sdr-notes #7: Sdr-transceiver not found and wspr question

karl-heinz-hi, 2019-06-04:
Hi Pavel,
at first many thanks for your excellent projects.
Mostly I use on the 14bit RP the wspr- transceiver and the sdr- transceiver
with my homebrew sdr-app based on Winrad.

I've just received the 16bit Red Pitaya. Looking to the stemlab-sdr-notes 
there is no  sdr-transceiver project  (as in the red-pitaya-notes).
Can I use the sdr-transceiver- files from the red-pitaya-notes without changes on the 
16bit RP?
Do you plan to adapt the transmitter/ extiIO to 16bit with e.g. higher sample rate possibly ...
that would be great.

WSPR-transceiver, Transmitting:
Can I start 'sessions' on each channel without problems , to get two
transmissions simultaneously?

Please advise.
best regards Karl

pavel-demin, 2019-06-04:
Hi Karl,

Thank you for your kind words.

My applications for the 14-bit Red Pitaya board can't be used on the new 16-bit board. The new 16-bit board has a different FPGA unit and a different ADC clock frequency. So, the applications should be adapted to these differences.

At the moment, I'm not going to port the SDR transceiver application to the new 16-bit board. From the feedback that I receive for this application, I understand that configuring it as two independent receivers and two independent transmitters isn't a good idea. So, a similar application for the new 16-bit board should be completely rewritten and it should contain many synchronous receivers and two synchronous transmitters.

The WSPR transceiver has only one transmitter. So, it only supports one transmission at a time.

Best regards,

Pavel

karl-heinz-hi, 2019-06-05:
Hi Pavel,
many thanks for your quick reply.
What a pity that no porting is planned up to now.
For the RP users via extIO.dll (as me) there is then no way to obtain the advantage of the 16bit RP. 
For us it's not important to have multiple transmitters/ receivers I think, only one receiver is sufficient.
Perhaps is it possible for you to create a extIO.dll for the your hpsdr- receiver ...
best regards
Karl
PS: the hope dies last

pavel-demin, 2019-06-05:
> Perhaps is it possible for you to create a extIO.dll for the your hpsdr-

It already exists:

https://github.com/IW0HDV/extio-iw0hdv/releases

I think that it's better to use it with SDR **transceiver** compatible with HPSDR.

pavel-demin, 2019-06-05:
Here is a direct link to the extio dll:

https://github.com/IW0HDV/extio-iw0hdv/releases/download/v1.0.5/extio_hpsdr_cmake_mgw-v1.0.5.dll

closed 2019-06-05 by pavel-demin