pavel-demin/stemlab-sdr-notes #8: no joy firing up the 16 bit Red Pitaya

BeauWebber, 2019-06-15:
Hmm, no joy firing up the 16 bit Red Pitaya.
Green LED OK, No Heartbeat LED.
BT Hub does not see it.
SD card version.txt and file dates identical to an SD card from a recent 14 bit Red Pitaya.

pavel-demin, 2019-06-15:
Since you don't provide any information about the version, I suppose that you're using the latest version ([stemlab-sdr-alpine-3.9-armv7-20190411.zip](https://github.com/pavel-demin/stemlab-sdr-notes/releases/download/20190411/stemlab-sdr-alpine-3.9-armv7-20190411.zip)).

> No Heartbeat LED.

There is no heartbeat LED configured in my SD card images.

By default, the LED blinker application starts at boot and the very first yellow LED (LED0) starts blinking.

> SD card version.txt and file dates identical to an SD card from a recent 14 bit Red Pitaya.

As far as I know, there is no file called version.txt in [stemlab-sdr-alpine-3.9-armv7-20190411.zip](https://github.com/pavel-demin/stemlab-sdr-notes/releases/download/20190411/stemlab-sdr-alpine-3.9-armv7-20190411.zip) and the file dates can't be identical to a different version of a different SD card image.

Could you please confirm that you're using [stemlab-sdr-alpine-3.9-armv7-20190411.zip](https://github.com/pavel-demin/stemlab-sdr-notes/releases/download/20190411/stemlab-sdr-alpine-3.9-armv7-20190411.zip) and that you're following the instructions from [this link](https://pavel-demin.github.io/stemlab-sdr-notes/alpine/)?

BeauWebber, 2019-06-15:
My apologies, I was just using the SD card that came with the Red Pitaya.
I will download your Alpine zip file now, But it may be a few days before I can try it.

pavel-demin, 2019-06-15:
If version.txt and file dates are identical to an SD card for the 14-bit boards, then it looks the 16-bit boards come with incompatible SD cards. The same problem is reported at [this link](http://dayton.contesting.com/pipermail/skimmertalk/2019-June/002520.html).

I see that the Red Pitaya developers provide a beta version of an SD card image for the 16-bit boards. It can be downloaded from [this link](http://downloads.redpitaya.com/downloads/STEMlab-122-16).

Since the issue is unrelated to the code from this repository, I'm closing it.

closed 2019-06-15 by pavel-demin