pavel-demin/stemlab-sdr-notes #9: sdr_receiver_hpsdr app and gr-hpsdr work unstable together

ashindin, 2019-06-25:
I cannot achieve stable behavior of these two applications together (sdr_receiver_hpsdr app and gr-hpsdr).
I'm using the following simple flow graph (gnuradio 3.7.11)
[image]

I can successfully see the spectrum only about every third run.

After each start I see in the logs:

> AlexFwdPwr =  0,0  AlexRevPwr =  0,0   ADCOver: 0  HermesVersion: 25 (dec)  19 (hex)
> AlexFwdPwr =  0,0  AlexRevPwr =  0,0   ADCOver: 0  HermesVersion: 25 (dec)  19 (hex)
> ...
> 
> LostRxBufCount = 0  TotalRxBufCount = 71838  LostTxBufCount = 0  TotalTxBufCount = 71838  CorruptRxCount = 0  LostEthernetRx = 378540

The same behavior I can see if I set up more than one receivers.

Sometimes I see only a small part of spectrum at the edges.
Is "LostEthernetRx" means lost packets? I have NetSDR receiver and it works correctly in my 100 Mbit network with a bandwidth 1 MHz.

Is it possible to perform the independent reception at the two RF inputs. 
Minimal example: the first receiver connected to IN1 with frequency f1 and the second receiver connected to IN2 with frequency f2 not equal f1.

pavel-demin, 2019-06-25:
It's possible that the problem is caused by the additional HPSDR server that I added to this application for STEMlab 122.88-16 SDR.

I'd suggest to try one of the following workarounds:
- use sdr_transceiver_hpsdr
- disable the additional HPSDR server by removing lines [9-12](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_receiver_hpsdr/app/start.sh#L9-L12) and [15](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_receiver_hpsdr/app/start.sh#L15) from [sdr_receiver_hpsdr/start.sh](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_receiver_hpsdr/app/start.sh).

ashindin, 2019-06-27:
> disable the additional HPSDR server by removing lines 9-12 and 15 from sdr_receiver_hpsdr/start.sh.  

Additionally I had to comment line 7 in sdr_receiver_hpsdr/stop.sh.
But all this helped only with lost packages. Now in the logs LostEthernetRx=0 all the time.

But strange client (gr-hpsdr) behavior remained.
For example at the first run of the sdr_transceiver_hpsdr I get the blank spectra:

[image]

At the next run I saw only one spectrum:
[image]
Only at the third-forth run I saw all three spectra:
[image]
At The next runs I saw nothing again.

The same behavior I get with sdr_receiver_hpsdr. The strange moment is that with the same flow graph the spectrum with central frequency 14996 kHz differ from that in sdr_transceiver_hpsdr:

[image]

Maybe all these are gr-hpsdr problems.

Another important problem that I don't know how to use the second RF input. 
For all values of the "Alex RX antenna" parameter, I see spectra only from the first RF input.
As I said earlier, I would like to have at least one receiver connected to IN1 and another receiver connected to IN2. Is it possible?

pavel-demin, 2019-10-25:
I'm closing this issue because I don't think that I can do anything to improve communication with gr-hpsdr.

closed 2019-10-25 by pavel-demin

pavel-demin, 2019-10-25:
> I would like to have at least one receiver connected to IN1 and another receiver connected to IN2. Is it possible?

The inputs can be configured by modifying the [start.sh](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_receiver_hpsdr/app/start.sh#L14) script.

The sdr-receiver-hpsdr program expects nine command line arguments:
```
sdr-receiver-hpsdr eth0 1 1 1 1 1 1 1 1
```
The first argument is the network interface, the remaining eight arguments are the inputs of the eight receivers (RX1, RX2, etc), where 1 corresponds to IN1 and 2 corresponds to IN2.