pavel-demin/stemlab-sdr-notes #10: Skimming for JT9?

fodonnell, 2019-07-07:
Hi Pavel, I recently bought two Red Pitaya 122.88-16 SDRs and have been getting excellent results with them configured as FT8 and WSPR skimmers, using your software. Thanks very much for the time and effort involved in creating this.

Can the FT8 software be modified so that it skims for JT9?

pavel-demin, 2019-10-25:
I think that it should be relatively easy to modify the FPGA configuration of the [multiband FT8 transceiver](https://pavel-demin.github.io/stemlab-sdr-notes/sdr-transceiver-ft8/), the [write-c2-files.c](https://github.com/pavel-demin/stemlab-sdr-notes/tree/master/projects/sdr_transceiver_ft8/app/write-c2-files.c) program and the [decode-ft8.sh](https://github.com/pavel-demin/stemlab-sdr-notes/tree/master/projects/sdr_transceiver_ft8/app/decode-ft8.sh) to register the correct number of samples with the correct sample rate.

However, finding/developing a stand-alone JT9 decoder is a bit more difficult.

According to the [PSK reporter statistics](https://www.pskreporter.info/cgi-bin/pskstats.pl), the are very few operators using the JT9 mode. Looks like this mode isn't very popular at the moment and there isn't much to skim.

fodonnell, 2019-10-25:
Thanks for the comments. I'm interested in JT9 because it's the primary mode for QSOs on the 630m band, at least here in the United States.

Since posting the request above, I put together a skimmer consisting of an Airspy HF+ SDR with Gqrx and Wsjtx software, both of which run on a Raspberry Pi 4. It's working fine for 630m JT9, although the advantage of using a Red Pitaya is that it would involve less hardware.