pavel-demin/stemlab-sdr-notes #11: wspr problems

dxhunter, 2019-07-08:
Dear Pavel
Wspr diversity mode causes problems on many bands.
The log shows twice same callsigns and same signalstrength per time-slot every 2Min.
The problem appears on 40m, 20m, 17m, 10m,
no data on 12m and 15m
On 600m only 1 channel works (could be my antenna).

Thank you very much for your help
73s de Eric

addendum
part of my log-file, many others possible
one of my write-c2-files.cfg

// frequency correction, from -100.0 ppm to +100.0 ppm
corr = 0.3;

// comma separated list of bands
// trailing commas are not allowed
bands = (
//  { freq  =  0.137500; chan = 1; },
  { freq  =  0.475700; chan = 1; },
  { freq  =  0.475700; chan = 2; },
  { freq  =  1.838100; chan = 1; },
  { freq  =  1.838100; chan = 2; },
  { freq  =  3.594100; chan = 1; },
  { freq  =  3.594100; chan = 2; },
  { freq  =  3.570100; chan = 1; },
  { freq  =  3.570100; chan = 2; },
  { freq  =  5.288700; chan = 1; },
  { freq  =  5.288700; chan = 2; },
  { freq  =  5.366200; chan = 1; },
  { freq  =  5.366200; chan = 2; },
  { freq  =  7.040100; chan = 1; },
  { freq  =  7.040100; chan = 2; }

//  { freq  = 14.097100; chan = 1; },
//  { freq  =  5.366200; chan = 1; }
//  { freq  = 24.926100; chan = 1; },
//  { freq  = 28.126100; chan = 1; },
//  { freq  = 50.294500; chan = 1; }
);

<DecodeFinished>
<DecodeFinished>
1122  -18  0.2   7.040027  0  2E0DSS IO82 33 

1122   -7  2.5   7.040057  0  DL7CHR JO62 23 

1122  -16  0.4   7.040094  0  G4ZAL IO80 23 

1122   -9  0.1   7.040102  0  DK6RS JN68 13 
1122  -10  0.1   7.040109 -1  <PA0ANH> JO22LH 27 
1122  -17  0.2   7.040119  0  G0CWT IO82 23 
1122    7  0.7   7.040124  0  F5VNE JN33 37 
1122  -19  1.6   7.040127  0  PA3DAI JO22 23 
<DecodeFinished>
<DecodeFinished>
1122   -18  0.2   7.040027  0  2E0DSS IO82 33 
1122   -7  2.5   7.040057  0  DL7CHR JO62 23 
1122  -16  0.4   7.040094  0  G4ZAL IO80 23 
1122  -9  0.1   7.040102  0  DK6RS JN68 13 
1122  -10  0.1   7.040109 -1  <PA0ANH> JO22LH 27 
1122  -17  0.2   7.040119  0  G0CWT IO82 23 
1122    7  0.7   7.040124  0  F5VNE JN33 37 
1122  -19  1.6   7.040127  0  PA3DAI JO22 23 
<DecodeFinished>

pavel-demin, 2019-07-08:
Dear Eric,

Thanks for the configuration file and for the logs.

I'd say that the behavior of the WSPR application that you observe is OK. There is no special diversity algorithm implemented in the WSPR application.

If you have similar antennas connected to both inputs of the STEMlab board, then it's quite normal that the decodes from both inputs have similar SNR values.

The commands on [line 63 in decode-wspr.sh](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_transceiver_wspr/app/decode-wspr.sh#L63) select the decodes with highest SNR. So, the duplicated decodes aren't uploaded to WSPRnet.

Best regards,

Pavel

dxhunter, 2019-07-08:
Hello Pavel

Thank you for your quick reply.
After I read your message the following test were done.
No change of the config-file at all.
Channel 2 antenna was disconnected. The result was the same as described in my former messages.
There was no change in the log compared to using 2 antennas, except 80m, there decodes now only from channel 1).
You could get the impression that 2 antennas were still connected.

Channel 1 was disconnected and channel 2 reconnected. Nearly no decodes now. Except the 80m spots from channel 2.

73s de Eric

dxhunter, 2019-07-16:
Is no one having the same problem?

pavel-demin, 2019-07-16:
I've just reread your previous comments and rechecked the code. Looks like I forgot to update the `sel` and `value` variables in [write-c2-files.c](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_transceiver_wspr/app/write-c2-files.c). These variables should be declared as `uint16_t`. The FT8 application has the same problem.

In the current version, only the first eight bands have configurable inputs.

dxhunter, 2019-08-01:
Dear Pavel
Thank you for your suggestions.
My knowledge on linux is pretty poor. Unfortunately I was not able to compile the changed file.
Therefore further help is necessary.

73s de Eric

pavel-demin, 2019-08-02:
Dear Eric,

Could you please test [this prerelease version](https://www.dropbox.com/sh/a1yecde4hs91ji0/AADxVwHWZ9U1VCu2eDtCOOp3a/stemlab-sdr-alpine-3.9-armv7-20190802.zip?dl=1)?

Best regards,

Pavel

dxhunter, 2019-08-04:
Dear Pavel
I have been using the new version for more than 24h. Everything is ok.
Thank you very much for fixing it.
Wspr is a great tool. One of its major features is the ability to compare antennas. Unfortunately in the decode-wspr.log you can't predict which port came the  report from. I wonder whether that can be changed?
73 de Eric

pavel-demin, 2019-08-04:
Thank you for testing the new version.

> Unfortunately in the decode-wspr.log you can't predict which port came the report from. I wonder whether that can be changed?

Since the reports are generated by the WSPR decoder, I think that the only way to add the antenna input numbers to the reports is to modify the WSPR decoder. The corresponding code is on line 1380 in wsprd.c:
https://sourceforge.net/p/wsjt/wsjt/HEAD/tree/wsjtx/trunk/lib/wsprd/wsprd.c#l1380

pavel-demin, 2019-10-25:
The fixed code is now in the latest release. Closing this issue.

closed 2019-10-25 by pavel-demin