pavel-demin/stemlab-sdr-notes #12: RF Output Level on CW Mode

Koichi-Koyama, 2019-07-08:
Dear Pavel-san, 
 
This is Koichi Koyama (Mr) from Japan, and I have been using  “OpenHPSDR mRX PS v3.4.9” for my new Red Pitaya SDR (16). 
OpenHPSDR is very nice software and thank you for your efforts for the development. 
 
After replacement of STEMLab125-14 with the new STEMLab122-88-16 SDR in my homebrew 200W HF transceiver, I have noticed that the output level of single carrier by CW key-down is increased by approx. 3dB or more from “Tune” mode.   
In case of STEMLab125-14, the output level of single carrier is same between CW key-down and clicking “Tune”. 
This is serious issue for me as it causes shutdown or breakdown of my linear amplifier due to excessing driving power.  

I have asked Red Pitaya technical staff who has advised me to contact you or open the issue in this forum, as it seems it is related to OpenHPSDR software. 

I would be grateful if you could advise me if this issue can be solved or any solution for me to take, other than changing driving level manually. 

Kind regards,

Koichi Koyama
7N1ICA

pavel-demin, 2019-07-09:
Thank you for testing the CW mode. I think that this problem should be fixed in the code with [this commit](https://github.com/pavel-demin/stemlab-sdr-notes/commit/c8d18d25c0b28e5d2a62a333741952631a662f46#diff-5f93faa912cb2f7a72e55d1e6dd8113e). The fixed code will be in the next release. Meanwhile, could you please check if [this prerelease version](https://www.dropbox.com/sh/a1yecde4hs91ji0/AABkdF8vSaSMIdzSCE5wk2oqa/stemlab-sdr-alpine-3.9-armv7-20190414.zip?dl=1) works better for you?

Koichi-Koyama, 2019-07-09:
Dear Pavel-san,

 

Thank you for very much your response. 

 

As far as I have checked the prerelease version, the RF output level on CW
mode has been fixed which is the same level of Tune mode. 

Do you have any idea when the new issue will be released?

 

Thank you again for your support.

 

Best regards,

 

Koichi Koyama

7N1ICA

<author@noreply.github.com>
(#12)

 

<https://github.com/pavel-demin/stemlab-sdr-notes/commit/c8d18d25c0b28e5d2a6
2a333741952631a662f46#diff-5f93faa912cb2f7a72e55d1e6dd8113e> . The fixed
<https://www.dropbox.com/sh/a1yecde4hs91ji0/AABkdF8vSaSMIdzSCE5wk2oqa/stemla
b-sdr-alpine-3.9-armv7-20190414.zip?dl=1>  works better for you?

-
<https://github.com/pavel-demin/stemlab-sdr-notes/issues/12?email_source=not
ifications&email_token=AMP2JGLRMK7YE7SMUVDWIATP6QRVZA5CNFSM4H63XVY2YY3PNVWWK
3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZPFPGI#issuecomment-5094
QRVZANCNFSM4H63XVYQ> .
<https://github.com/notifications/beacon/AMP2JGM45LGNDE7YKUIUFZ3P6QRVZA5CNFS
M4H63XVY2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODZPFPGI
.gif> 

 

  _____  

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient> 

Scanned by McAfee
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient>  and confirmed virus-free.

pavel-demin, 2019-10-25:
The fixed code is now in the latest release. Closing this issue.

closed 2019-10-25 by pavel-demin

Koichi-Koyama, 2019-10-28:
> 宛先: "pavel-demin/stemlab-sdr-notes" <stemlab-sdr-notes@noreply.github.com>

Thank you for the information and well noted. 

Koichi Koyama
7N1ICA

----- 元のメッセージ -----

送信済み: 2019年10月26日 午前 3:48