pavel-demin/stemlab-sdr-notes #13: How to format SD card?

Evidlo, 2019-07-21:
I have a Stemlab 125-10 and I'm trying to use [this image](https://github.com/pavel-demin/stemlab-sdr-notes/releases/download/20190411/stemlab-sdr-alpine-3.9-armv7-20190411.zip).

I've been following [this guide](https://pavel-demin.github.io/stemlab-sdr-notes/alpine/), but it just says to unzip and copy files and makes no mention of how to format the sdcard.  I've tried a single FAT32 partition and a single ext4 partition spanning the whole card, but neither have worked.

Maybe in the future it would be easier to provide an image that we could just `dd` directly to the sdcard?

pavel-demin, 2019-07-22:
The SD card images from this repository aren't compatible with 125-10. They work only with STEMlab 122.88-16 SDR.

Please use the SD card images from
https://github.com/pavel-demin/red-pitaya-notes/releases

> I've been following this guide, but it just says to unzip and copy files and makes no mention of how to format the sdcard.

As far as I know, most of the SD cards are formatted FAT32 out of the box. My instructions work with this kind of card.

> Maybe in the future it would be easier to provide an image that we could just dd directly to the sdcard?

No, it won't be easier. I believe, that copying a few files to a freshly unpacked SD card is the easiest solution.

pavel-demin, 2019-07-22:
Correction. As I don't have Red Pitaya 125-10, I'm not sure if the SD card images from
https://github.com/pavel-demin/red-pitaya-notes/releases
work with this type of board.

If they don't, then the only solution is to use the SD card images provided by the Red Pitaya developers and install my applications via the application marketplace.

closed 2019-07-22 by pavel-demin