pavel-demin/stemlab-sdr-notes #14: WSPR not transmitting

karl-heinz-hi, 2019-08-13:
Hello Pavel,
I've tried to use the transmitting part of the WSPR transceiver app.:
Editing the transmit-wspr-message.cfg and update the wspr.cron file  (by the way, last step is missing in your description). In the file wspr.cron the time parameter for TX was changed by you from '1-59/2' (in the 14 bit) to '1-51/10'.... Why?. 
I have used '1-59/2' to get 2 minutes intervals and started the app via web frontent.
**The log file says all OK:**
Tue Aug 13 10:16:01 UTC 2019
Transmitting ...
Tue Aug 13 10:17:01 UTC 2019
Sleeping ...
Tue Aug 13 10:18:01 UTC 2019
Transmitting ...
But there is no output signal. Is there an additional step required?
Using: 16 bit Stemlab, SW = 'stemlab-sdr-alpine-3.9-armv7-20190411' and also the '0802' version.
To assure the there is no HW fault I've used the generator from the red pitaya image = Output ok.
Please advise.
best regards Karl-Heinz / dd9ok

pavel-demin, 2019-08-13:
Thank you for testing the WSPR transceiver application.

Looks like I forgot to update the [lines 113-114 in transmit-wspr-message.c](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_transceiver_wspr/app/transmit-wspr-message.c#L113-L114) when I added eight receivers. These lines should should be changed to
```
  coef[0] = ((uint16_t *)(cfg + 72));
  coef[1] = ((uint16_t *)(cfg + 74));
```

>  In the file wspr.cron the time parameter for TX was changed by you from '1-59/2' (in the 14 bit) to '1-51/10'.... Why?.

I don't remember changing this line in wspr.cron. I've just checked both versions and they are the same. Here are the links to both versions of wspr.cron:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/wspr.cron
https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_transceiver_wspr/app/wspr.cron

karl-heinz-hi, 2019-08-13:
Seems that I've compared my edited wspr.cron of the 14bit environmet with the current 16 bit file. 
My mistake. Sorry about that...

karl-heinz-hi, 2019-08-13:
Pavel, I hope you can help me possibly.
I'm going to use the UART/ serial port. Therefore, a serial port extension for python 3 is required.
I could find only pyserial working. Installation is done with:
  python3 -m pip install pyserial
  lbu commit -d
Everything works fine, but after reboot 'import serial' fails. My workarround is installing pyserial with a start.sh during booting, but that is not the correct way.
Any ideas?
Many thanks in advance

pavel-demin, 2019-08-13:
I think that `pip install --user` should work better.

karl-heinz-hi, 2019-08-13:
Many thanks Pavel... that is the solution...
python3 -m pip install --user pyserial

karl-heinz-hi, 2019-08-21:
Hello Pavel,
can you please generate a new (prerelease) version that can transmit, because is not possible for me.
...that would be great...

pavel-demin, 2019-10-23:
This problem is now fixed in the code with [this commit](https://github.com/pavel-demin/stemlab-sdr-notes/commit/988dd6286aac235de5dad2d22ac691faf769ce4a).

Here is a link to a pre-release version of the SD card image containing the new code:

https://www.dropbox.com/sh/a1yecde4hs91ji0/AAD9e3u847NM2UU0pGpzAPiAa/stemlab-sdr-alpine-3.9-armv7-20191023.zip?dl=1

pavel-demin, 2019-10-25:
The fixed code is now in the latest release. Closing this issue.

closed 2019-10-25 by pavel-demin