pavel-demin/stemlab-sdr-notes #15: Adaptation "SDR transceiver" and "Wideband SDR transceiver" applications to Stemlab 122.88-16

ashindin, 2019-10-27:
Is it difficult to implement?

pavel-demin, 2019-10-27:
It isn't very difficult to port the SDR transceiver application to the STEMlab 122.88-16 SDR.

I don't see any reason to port the **wideband** SDR transceiver application to the STEMlab 122.88-16 SDR. Transmitting and receiving complex baseband signals won't work with the STEMlab 122.88-16 SDR because of the input and output transformers and the STEMlab 122.88-16 SDR has enough resources to run the SDR transceiver application at higher sample rates.

ashindin, 2019-10-28:
Did I understand correctly, that STEMlab 122.88-16 SDR has **not** enough resources?
If yes, could you briefly explain the fundamental difference between STEMlab 125-14 and STEMlab 122-16 (except sample rate and ADC resolution)?

pavel-demin, 2019-10-28:
> Did I understand correctly, that STEMlab 122.88-16 SDR has not enough resources?

I'm afraid that you didn't understand correctly. I should have split the last sentence of my previous comment into shorter pieces.

The **wideband** SDR transceiver application has the following two features:
 - transmitting and receiving complex baseband signals
 - higher sample rates

The first feature won't work with the STEMlab 122.88-16 SDR because of the input and output transformers blocking low frequencies.

The second feature could be integrated with the SDR transceiver application because the STEMlab 122.88-16 SDR has enough resources to run the SDR transceiver application at higher sampling rates.

That's why I currently don't see any reason to port the **wideband** SDR transceiver application to the STEMlab 122.88-16 SDR. 

Do you need some other features of  the **wideband** SDR transceiver to be ported to the STEMlab 122.88-16 SDR?

pavel-demin, 2019-10-28:
> could you briefly explain the fundamental difference between STEMlab 125-14 and STEMlab 122-16 (except sample rate and ADC resolution)?

These two boards have different ZYNQ chips: 125-14 has Z-7010 and 122-16 has Z-7020. The difference between the chips can be found in the product table at [this link](https://www.xilinx.com/products/silicon-devices/soc/zynq-7000.html#productTable).

Here is a screenshot showing the FPGA resources available on the Z-7010 and Z-7020 chips:
[image]

The Z-7020 chip on the STEMlab 122.88-16 SDR has 2-3 time more FPGA resources.

ledodd, 2019-10-28:
Pavel-Demin
I am very interested in this thread. I monitor (receive only) 15 to 30 MHz a 15 MHz instantaneous bandwidth for radio astronomy. How can I do this with 122.88-16?
Thank you so much. 
Larry

pavel-demin, 2019-10-28:
> I monitor (receive only) 15 to 30 MHz a 15 MHz instantaneous bandwidth for radio astronomy. How can I do this with 122.88-16?

Thanks for your interest. However, I'm not sure if I understand how this is related to this thread. The SDR transceiver application has much lower bandwidth. I think that when ported to the STEMlab 122.88-16 SDR, the maximum sample rate will be 1.92 MSPS with the usable bandwidth of about 1.6-1.8 MHz.

ledodd, 2019-10-28:
> <stemlab-sdr-notes@noreply.github.com>
> <comment@noreply.github.com>
> transceiver" and "Wideband SDR transceiver" applications to Stemlab
> 122.88-16 (#15)

Pavel
Yes that is what I needed to know.  Yes I see about 1.6 to 1.8 bandwidth 
so I am seeing all that this application can provide.  I needed that 
confirmation. Thanks!  I hope someone will make a new spectrum analyzer 
application for 122.88-16.
Thanks again.
Larry

F6ITU, 2019-10-28:
Hi Pavel
Sorry for hijacking this tread, but what would be the procedure to add a "feature request" ?

I'm working on a HPSDR frontend clone that could probably allow the "Wideband sdr feature" on the 16 bits platform. 

In other words, a more classical "Attenuator/ADC driver/Antialiasing filter" in the RX path and "lpf/low power driver" in the TX path, _without any_ transformer, at least in the RX path

The feature request is limited to the gateware integration of the attenuator's command (DAT31sp+, strictly the same than the one used in the hpsdr familly) 

A prototype series is planned for December.. this year :-  )) 

Thnk
Marc

F6ITU, 2019-10-28:
Tnks
I was to shy to consider it as an issue :- )) 
I'll work on it next week
Tnks
Marc

ashindin, 2019-10-29:
> Pavel Yes that is what I needed to know. Yes I see about 1.6 to 1.8 bandwidth so I am seeing all that this application can provide. I needed that confirmation. Thanks! I hope someone will make a new spectrum analyzer application for 122.88-16. Thanks again. Larry  

Larry, did you see this project?
[https://github.com/tknopp/RedPitayaDAQServer](https://github.com/tknopp/RedPitayaDAQServer)
It is not yet compatible with 122.88-16, but with 125-14 allows to "continuously acquire the signals of the fast ADCs at a decimation of up to 8 (15.625 MSamples/s)."

Alexey

ashindin, 2019-10-29:
> > could you briefly explain the fundamental difference between STEMlab 125-14 and STEMlab 122-16 (except sample rate and ADC resolution)?
> 
> These two boards have different ZYNQ chips: 125-14 has Z-7010 and 122-16 has Z-7020. The difference between the chips can be found in the product table at [this link](https://www.xilinx.com/products/silicon-devices/soc/zynq-7000.html#productTable).
> 
> Here is a screenshot showing the FPGA resources available on the Z-7010 and Z-7020 chips:
> [image]
> 
> The Z-7020 chip on the STEMlab 122.88-16 SDR has 2-3 times more FPGA resources.

Thanks for explanation, Pavel.

I forgot about more powerful FPGA (I knew about it).
Where I can find more information about "input and output transformers blocking low frequencies" in 122.88-16?

Do you mean that 125-14 has no such transformers?

Is it possible (in principle) to use 122.88-16 to generate (and transmit) low duty-cycle radio pulses (for example: duration -- 40-400 usec, period -- 5-25 msec and filling frequency about 5 MHz?

Regarding transmitting I'm searching for ability to transmit such pulses.
Regarding receiving I'm looking for ability to receive (continuous sample with 16 bit resolution and store) at least two independent signals with different central frequencies and sample rate just about 1.6-1.8 MHz.

Alexey

ledodd, 2019-10-29:
Alexey
No was not aware of that application. I will check it out. Thank you. 
Larry

pavel-demin, 2020-02-28:
> Is it possible (in principle) to use 122.88-16 to generate (and transmit) low duty-cycle radio pulses (for example: duration -- 40-400 usec, period -- 5-25 msec and filling frequency about 5 MHz?

If you're still looking for a solution, then I've just realized that the [Pulsed NMR project](http://pavel-demin.github.io/red-pitaya-notes/pulsed-nmr) does something similar and it's already ported to STEMlab 122.88-16 SDR:
https://github.com/pavel-demin/stemlab-sdr-notes/tree/master/projects/pulsed_nmr

ledodd, 2020-02-28:
Thanks!  I’ll check it out. 
Larry

pavel-demin, 2020-02-28:
> Thanks!  I’ll check it out.

I was answering Alexey's question.

ashindin, 2020-02-28:
> If you're still looking for a solution, then I've just realized that the [Pulsed NMR project](http://pavel-demin.github.io/red-pitaya-notes/pulsed-nmr) does something similar and it's already ported [...]

Thank you, Pavel! I will try it.
At the moment, our team is working on the implementation of a pulse generator for the ionosonde with STEMlab 122.88-16. Current step is to make a changing in the carrier frequency from pulse to pulse.

pavel-demin, 2020-02-28:
> Current step is to make a changing in the carrier frequency from pulse to pulse.

In the current version of the pulsed_nmr project, it's possible to set the phase offset from pulse to pulse. Each pulse has four parameters: delay before, pulse duration, delay after, phase offset. The phase offset can be easily replaced with the phase increment (carrier frequency).