pavel-demin/stemlab-sdr-notes #16: very wideband DDC/s (copied from pavel-demin/red-pitaya-notes)

johnretired, 2019-12-22:
Can the STEMlab 122.88-16 SDR support TWO wideband DDC/s with a 6.144 MHz bandwidth each? I don't need any DUC/s implemented on the device.

pavel-demin: If 16-bit I and Q values are OK for your project, then I think that it's possible. The bottleneck is the network bandwidth.

Is the network bandwidth limitation due to the STEMlab not being able to push the data out the GbE?  In my previous life I was using an acquisition device that was able to push 3 x 5.12 MHz (16bit I & Q, 4 bytes total per complex sample) bandwidth streams out a single GbE line without any issues.  Just trying to understand where the bottleneck is.

pavel-demin, 2019-12-22:
Thanks for moving this issue to this repository.

> Is the network bandwidth limitation due to the STEMlab not being able to push the data out the GbE?

I don't know what is the maximum achievable network bandwidth of the STEMlab 122.88-16 SDR.

I once used the [adc_test](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/adc_test) project to test how fast the ADC samples could be transferred to a remote PC. Here is a link to a post on the Red Pitaya forum with the results of the tests:

http://forum.redpitaya.com/viewtopic.php?t=317&p=2118#p2118

Transferring two bytes per sample, the highest  sample rate that I achieved was 31.25 MSPS. With the client program running under Windows 7, the Windows 7 network monitoring showed 63-65 MB/s network usage.

31.25 MSPS with two bytes per sample corresponds to 477 Mbit/s (31.25e6 * 16 / 2^20).

So, I know that 477 Mbit/s is achievable with my current programming techniques.

Two DDCs generating 6.144 MSPS require
- 375 Mbit/s (6.144e6 * 32 * 2 / 2^20) for 16-bit I and Q samples
- 750 Mbit/s (6.144e6 * 64 * 2 / 2^20) for 32-bit I and Q samples

I think that 375 Mbit/s should be OK for the STEMlab 122.88-16 SDR. However, I'm not so sure about 750 Mbit/s .

Are 16-bit I and Q values OK for your project?

johnretired, 2019-12-22:
16-bit I and Q values are OK.

pavel-demin, 2019-12-22:
I think that even 7.68 MSPS should be possible with 16-bit I and Q values.

johnretired, 2019-12-22:
Thanks for the information.  I’ve been very hesitate to purchase a device unless I was confident that it could support 2 x 6.144 MHz bandwidth (16-bit I and Q samples) streams to a Linux PC.  I plan on proceeding with the purchase of the STEMlab 122.88-16 SDR within the next few days.  I have a lot of C/C++ experience both with TI DSP devices and Host based but none with FPGA devices.   What I want to be able to do is set the Center Frequencies of the two DDC/s at startup and enable the output streaming to a Linux PC. The DDC’s 6.144 MHz bandwidth can be fixed.  Are there any STEMlab 122.88-16 SDR applications close to what I’m trying to do available?

pavel-demin, 2019-12-22:
I've just put together a project called [sdr_receiver_wide](https://github.com/pavel-demin/stemlab-sdr-notes/tree/master/projects/sdr_receiver_wide). I think it has all the components you're looking for.

johnretired, 2019-12-27:
I've received my STEMlab 122.88-16 SDR and got it up and running.  Is there an example of how to implement the "sdr_receiver_wide" project code?

pavel-demin, 2019-12-27:
The code is mostly implemented. It just needs to be built, copied to the board and tested. Most of the building instructions can be found in the [Development machine](https://pavel-demin.github.io/stemlab-sdr-notes/development-machine/) and [LED blinker](https://pavel-demin.github.io/stemlab-sdr-notes/led-blinker/).

If you have Vitis Core Development Kit 2019.2 installed, then the FPGA configuration and the server code can be built with the following commands:
```
source /opt/Xilinx/Vitis/2019.2/settings64.sh

git clone https://github.com/pavel-demin/stemlab-sdr-notes
cd stemlab-sdr-notes

make NAME=sdr_receiver_wide bit

arm-linux-gnueabihf-gcc -static -O3 -march=armv7-a -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard -D_GNU_SOURCE projects/sdr_receiver_wide/server/sdr-receiver-wide.c -o sdr-receiver-wide -lm
```

johnretired, 2019-12-27:
Thanks.  I got it all up and going.  Just a few of questions:
I assume this is the center frequency for channel #1?
*(uint32_t *)(cfg + 4) = (uint32_t)floor(600000 / 122.88 * (1<<30) + 0.5);
I assume this is the center frequency for channel #2?
*(uint32_t *)(cfg + 8) = (uint32_t)floor(600000 / 122.88 * (1<<30) + 0.5);
I'm not clear on how you're identifying which channel the data is coming from, channel #1 or channel #2?

pavel-demin, 2019-12-27:
> I assume this is the center frequency for channel #1?
> *(uint32_t *)(cfg + 4) = (uint32_t)floor(600000 / 122.88 * (1<<30) + 0.5);
> I assume this is the center frequency for channel #2?
> *(uint32_t *)(cfg + 8) = (uint32_t)floor(600000 / 122.88 * (1<<30) + 0.5);

Correct. 600000 is the center frequency in Hz.

> I'm not clear on how you're identifying which channel the data is coming from, channel #1 or channel #2?

The I and Q samples of the two channels are stored in the form of the following C struct:
```
struct samples
{
  int16_t I1;
  int16_t Q1;
  int16_t I2;
  int16_t Q2;
}
```

johnretired, 2019-12-30:
Two DDCs generating 6.144 MSPS require  375 Mbit/s (6.144e6 * 32 * 2 / 2^20) for 16-bit I and Q samples. 375 Mbit/s / 8 bits = 46.875 MB per sec.  I assume this is about what the FPGA should be providing.  On average I'm only seeing about 30 MB per sec being read by the client on my host (12% cpu load).  Running top on the Alpine OS (4.14.150-xilinx) has about 50% for processor zero and about 45% for processor one when running sdr-receiver-wide.  It appears that the sdr-receiver-wide isn't able to send the data fast enough through the socket.  I'm assuming I must not have something setup correctly.

johnretired, 2019-12-31:
Additionally, I have a direct network cable connection from the host PC to the Board.  I also tried the application on the STEMlab_122-16_OS_0.98-5_stable.img and the throughput was about the same.

johnretired, 2019-12-31:
I changed the sysctl wmem socket buffer sizes on the board and the sysctl rmem socket buffer sizes on my host PC but it didn't make difference in the throughput.

pavel-demin, 2020-01-01:
Thank you for testing the new sdr_receiver_wide project and finding the problem with the network transfer rate.

I tested several SD card images and found that the problem only occurs with the 2019 releases.

In the beginning of 2019, I started using Alpine Linux 3.9 and the armv7 architecture introduced with that version.  Normally, armv7 packages should be better optimized for the Cortex-A9 processor than armhf packages. However, our tests show that this isn't always the case.

Switching back to the armhf architecture solves the problem with the network transfer rate.

Here is a link to a pre-release version of the SD card image that I've just built using the armhf packages:
https://www.dropbox.com/sh/a1yecde4hs91ji0/AABShs7dvAfHUZfoCIMvQQsqa/stemlab-sdr-alpine-3.9-armhf-20200101.zip?dl=1

This SD card image contains the files required to run the new sdr_receiver_wide application. Here are the commands to configure the FPGA and start the server program:
```
cd apps/sdr_receiver_wide
cat sdr_receiver_wide.bit > /dev/xdevcfg
./sdr-receiver-wide
```

johnretired, 2020-01-01:
Thank you.  It's been a couple of years since I retired and I haven't been doing any serious programming to speak of.  I was concerned about whether I had everything set up correctly.  But now the transfer rates are on average looking right for two 6.144 MHz streams.  I appreciate you taking the time to get the problem corrected.  Thanks again.

Another observation.  When running "mpstat -P ALL 1" I'm getting (example below):

12:56:18     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
12:56:19     all       16.24    0.00     42.64    0.00      0.00     8.12     0.00      0.00    32.99
12:56:19       0        0.00    0.00     84.00    0.00       0.00    16.00    0.00      0.00     0.00
12:56:19       1      32.99    0.00       0.00    0.00       0.00      0.00    0.00      0.00     67.01

I'm a little concerned about the 0.00 %idle time for CPU #0. I don't have a signal generator where I can verify the integrity of the acquired data.  I plan on getting a cheap one soon.  This is a hobby so I'm very low budget.

pavel-demin, 2020-01-01:
One of the RF outputs on the STEMlab 122.88-16 SDR board can be used as a signal generator. I can add some DDS blocks, multipliers and adders to generate single and dual tone sinusoidal signals.

pavel-demin, 2020-01-01:
BTW. When continuously running with high CPU usage the board requires active cooling.

Small 30x30mm 5V fans can be attached to the board's heatsink with two screws and powered from the onboard 5V fan connector.

I use the following two fans and they work well with my Red Pitaya boards:
http://uk.farnell.com/jsp/search/productdetail.jsp?id=1924848
http://uk.farnell.com/jsp/search/productdetail.jsp?id=1924852

With these fans attached to the heat sink, the main chip's temperature is around 40 degrees.

johnretired, 2020-01-01:
Being able to generate a single tone sinusoidal signal at a fixed frequency would work for me.  Would it be possible for me to turn it off in the tcl file when the signal isn't required for testing?  This way I could generate two bit files, one with it enabled and one with it not.  My only concern would be is this additional capability going to cause any performance issues with the existing DDC/s performance?

johnretired, 2020-01-01:
Could the amplitude of the generated single tone sinusoidal signal be adjustable?

johnretired, 2020-01-02:
By adjustable I don't mean it has to be interactive just that I change it in the server's c code and re-compile.

pavel-demin, 2020-01-02:
Here is a link to a new pre-release version of the SD card image with a simple signal generator:
https://www.dropbox.com/sh/a1yecde4hs91ji0/AABLiHJ6fDSE0DKp6bVKBiANa/stemlab-sdr-alpine-3.9-armhf-20200102.zip?dl=1

The following commands can be used to start the sdr_receiver_wide application and the generator:
```
cd apps/sdr_receiver_wide
./starts.sh
./gen 1 32766 1000000
```
The first parameter is the RF output number (1 or 2), the second parameter is the signal level (from 0 to 32766), the third parameter is the frequency in Hz.

Since the STEMlab 122.88-16 SDR doesn't have any on-board filter, you'll probably need an external low pass or band pass filter.

johnretired, 2020-01-02:
Thought I had some SMA cables but couldn't find them.  I'll order some and get back in touch next week.  Thanks.

johnretired, 2020-01-07:
An error occurred when I ran the command:
./start.sh
ip: can't find device 'mvl0'

Just wanted to pass this along.  The program appears to be running correctly.  My client did attach and is receiving data.  I'll check the data in a few.

johnretired, 2020-01-07:
About 550 is the most I can use for the signal level or it appears the data is overdriven.  I'm not convinced it's not something in my code reading the data, writing it to a file and then presenting it in a GUI.  Otherwise, everything seems to be working correctly.

johnretired, 2020-01-08:
I've been trying to figure out what's going on with my build process because when I use your pre-built bit file and executable it's socket output is near 6 mega-samples second/channel.  But when I build the bit file and executable it's output isn't close.  I've switched between bit files and executables and it's both that aren't as optimized as yours.  I'm building both on my host PC.

HOST PC:

arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (GCC) 8.2.0

arm-linux-gnueabihf-gcc -static -O3 -march=armv7-a -mcpu=cortex-a9 -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard sdr-receiver-wide.c -o sdr-receiver-wide -lm -lpthread

FPGA:

I created /home/opt to install the Xilinx tools

I installed both Xilinx_Vitis_2019.2_1106_2127.tar.gz & Xilinx_Vivado_Vitis_Update_2019.2.1_1205_0436.tar.gz

source /home/opt/Xilinx/Vitis/2019.2/settings64.sh

make NAME=sdr_receiver_wide bit
........
INFO: [Common 17-83] Releasing license: Implementation
56 Infos, 0 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:01:10 ; elapsed = 00:00:54 . Memory (MB): peak = 3207.371 ; gain = 495.180 ; free physical = 31895 ; free virtual = 57192
# close_project
INFO: [Common 17-206] Exiting Vivado at Wed Jan  8 14:30:06 2020...

johnretired, 2020-01-08:
Should have been # close_project

pavel-demin, 2020-01-08:
The pre-built executable is built by the [alpine.sh](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/scripts/alpine.sh) script using the compiler from the Alpine Linux distribution.

It's also possible to build it directly on the STEMlab 122.88-16 SDR with the following commands:
```
apk add make gcc

cd apps/sdr_receiver_wide
rw

make
```

pavel-demin, 2020-01-08:
I did a few tests and found that the single process server works well with the 6.144 MSPS rate. So I simplified the server code and made the sample rate configurable (from 0.96 to 7.68 MSPS).

Here is a link to a new pre-release version of the SD card image with a the new version of the sdr_receiver_wide application:
https://www.dropbox.com/sh/a1yecde4hs91ji0/AABwzWUrl6WZfAonbE2hJ7rJa/stemlab-sdr-alpine-3.9-armhf-20200108.zip?dl=1

The continuity of the received signal seems to be OK. Here is an I/Q waveform and a spectrum of ~30M samples with the RX frequency set to 10 MHz and the TX frequency set to 10 MHz + 1 Hz:
[image]
[image]

johnretired, 2020-01-08:
Thanks.  What version of the Xilinx tools are you using?

pavel-demin, 2020-01-08:
> What version of the Xilinx tools are you using?

Xilinx_Vitis_2019.2_1106_2127.tar.gz

pavel-demin, 2020-01-08:
The output of `make NAME=sdr_receiver_wide bit` from you previous comment looks OK. The bitstream file can found in the `tmp` directory (`tmp/sdr_receiver_wide.bit`).

pavel-demin, 2020-01-14:
There is now a new version with the following optimizations and improvements:
- improved bit usage in I/Q samples
- memcpy replaced with NEON instructions
- reduced buffer size to fit in the L2 cache
- commands to remotely change the sample rate and center frequencies

With this new version and with a good network connection, I can receive the I/Q samples at 7.68 MSPS.

Here is a link to a SD card image with this new version:
https://www.dropbox.com/sh/a1yecde4hs91ji0/AADbU_bh8XZCuqFzaSLlb2ota/stemlab-sdr-alpine-3.9-armhf-20200114.zip?dl=1

johnretired, 2020-01-15:
I jumped right to testing at the 7.68 MHz bandwidth.  I haven’t implemented the socket interface and just modified the code for a decimation setting of 8.  Everything seems to work fine for a while (many seconds) until the transfer rate would suddenly drop down from about 7.68 to a lesser value for a short period of time (several seconds) and then return back to about 7.68.  I brought up “mpstat -P ALL 1” on the redpitaya board and noticed when the transfer rate would drop on my host the percent usage on the redpitaya’s CPU #0 would drop and CPU #1 would go up for a short period of time and then return back to what I would call normal.  I’m not sure exactly what’s the OS is doing on the redpitaya board but in my past work with Linux I’ve noticed the OS wants to move high cpu usage processes around and when that happens usually the throughput suffers for a short period of time.  What I’ve done to fix this problem is to use cpu affinity.  I’m not sure if this is absolutely the problem or if the Alpine supports cpu affinity.  Just thought I would bring it up.

FYI: I have a direction network connection between the redpitaya board and my host.

pavel-demin, 2020-01-15:
Thank you for testing this new version.

It's possible to change the affinity and the priority of the process using the `sched_setaffinity` and `sched_setscheduler` functions. I've just added them to the server code with [this commit](https://github.com/pavel-demin/stemlab-sdr-notes/commit/3fa120f5c19adfb6f080daaa373b679eb3af8ae6).

Could you check if this improves performance?

johnretired, 2020-01-15:
I changed the decimation setting to 8 and recompiled and after 10 minutes I haven't seen any problems.  It looks like that was the issue.  For years I've use both cpu affinity and the real-time mode on all my host based DSP applications.  Thanks again for your quick response.  I've very happy with the product and your support.

johnretired, 2020-01-16:
I ran it for over an hour and didn't see the issue.

pavel-demin, 2020-01-16:
Thank you for the tests. I'm glad the new version works well now.

I also added a simple signal continuity test:
https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_receiver_wide/client/test-signal-continuity.c

So now I can test the continuity of the signal over long periods.

This program receives the I/Q samples and calculates the difference between a sample and a previous one. If the difference is greater than 50, it prints it on the screen. If it doesn't print anything, it means that the signal continuity is OK.

If the Ethernet port of the STEMlab 122.88-16 SDR board is directly connected to the Ethernet port of the computer using a good quality CAT6 cable, the test runs for many minutes without showing any sign of discontinuity.

johnretired, 2020-01-17:
I initially used 32766 for the gen amplitude but it was causing the application to print out error values.  I changed the gen amplitude to 16383 and ran it for over 3 hours without any error output.  The specifications for the STEMLAB 122-16 state the max output is 1Vpp/+4 dBm and the max input is 0.5Vpp/-2dBm.  I have a Mini-circuits SLP-30 between the output and input.  It's insertion loss below 30 MHz is less than 1 dB.  What gen amplitude were you using?

pavel-demin, 2020-01-17:
I forgot that I changed the input range on my STEMlab 122.88-16 SDR board and its input range is now 1 Vpp. Changing the position of the jumper connected to the SENSE pin of the ADC chip increases the dynamic range by 6 dB without decreasing the sensitivity. The modification is shown on the first page of [this PDF file](https://www.dropbox.com/sh/a1yecde4hs91ji0/AAAM9QNyKQ1n3b2zBLpKdJ7fa/test/RP16Bit_eng.pdf?dl=1).

So 32766 works with my board, but 16383 should be used with an unmodified board.

johnretired, 2020-01-17:
Thanks for the information.

johnretired, 2020-01-18:
Just went over 12 hours without any errors.