pavel-demin/stemlab-sdr-notes #17: Is this normal ? Power setting level of 255 renders a completely saturated spectrum

satfan52, 2020-01-08:
Hello Pavel,

During my tests with the 122.888 16 board (original - non modified) I noticed something that seemed to me abnormal, but maybe it isn't. Thanks for telling me if I should consider this as working as designed or not.  

When I set in QUISK the power level settings to 255/255 I get a clean spectra with 125-14 board (see the first picture below at 50 Mhz). The output power measured at 50 Mhz is 7-8 dBm which seems to me totally normal.

If I do the very same measurement (so power level of 255/255) but with the 122.88-16 I get a completely messed spectrum (see second picture also at 50 Mhz) 

Reducing the power level setting to 240/255 solves the issue but the output power is limited to about 0.5 dBm which is much lesser than the 5dBm expected according to the specifications of the redpitaya (see third picture also at 50 Mhz)

Regards (and happy new year)
Peter 

[image]

[image]

[image]

pavel-demin, 2020-01-10:
Hello Peter,

I can't reproduce this problem. I generated a single tone test signal at 50.5 MHz using PowerSDR mRX PS and the spectrum looked OK:
[image]

I checked the tx_level value on the server side and it was 255.

Best regards,

Pavel

pavel-demin, 2020-01-10:
> Reducing the power level setting to 240/255 solves the issue but the output power is limited to about 0.5 dBm which is much lesser than the 5dBm expected according to the specifications of the redpitaya (see third picture also at 50 Mhz)

With a 50 Ohm termination at the oscilloscope side, the amplitude of a single tone test signal is about 960 mVpp (3.6  dBm).

satfan52, 2020-01-11:
Ok, thanks, will try with powersdr also on my side

satfan52, 2020-01-11:
Ok, I tested with default settings of PowerSDR  and the "drive ruler " set to 100%, the spectra is clean and it is the very same than with QUISK and the "power setting level" set to 240/255. The power level measured at 50 Mhz and 50.5 Mhz with my JDSU spectrum analyzer is like with QUISK 0.5 dBm and this also what I have measured with my oscilloscope after compensating for the loss of the anti-alias filter that is necessary for making this measurement with the 122.88-16. Without any anti-alias filter, I measure 2.8 dBm with my wideband AD8307 power meter. 

So I think one can conclude that the issue (if any) is with QUISK. 

Regarding the power level, if your measurement of 3.6 dBm corresponds to the power emitted by the 125.88-16 over the whole spectra of the 122.88-16 then everything seems to me explainable, my wide bandwith of 2.8 dBm is close from your 3.6 dbm measurement and the two measurements are close of the 5 dBm specified by  Stemlab for the 122.88-16 board. However if  your 3.6 dBm is the actual power of the 50.5 Mhz component only, then there is still a mystery to be solved (on my side most likely)

pavel-demin, 2020-01-11:
I don't have a filter for 50 MHz. So I measure the amplitude without any filter.

It's possible that the amplitude of the test signal generated by QUISK is higher than the amplitude of the test signal generated by PowerSDR mRX PS.

The tx_level value is calculated on [line 1033 in sdr-transceiver-hpsdr.c](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L1033). Replacing 85.920 with a lower value could fix the problem when using the application with QUISK. Another solution would be to reduce the amplitude of the test signal in QUISK.

satfan52, 2020-01-11:
Ok clear, many thanks.  Pragmatically, I guess Quisk users should simply know that they should not exceed 240/255 in the power settings with the 122.88-16 board. For my own use, to avoid accidental mistakes, I may modify the parameter that you mention and then recompile. Thanks again for everything

closed 2020-01-11 by satfan52