pavel-demin/stemlab-sdr-notes #18: Embedded SDR transceiver for STEMLab 122-16

redpitaya0815, 2020-04-24:
Dear Pavel, do you plan a version of the Embedded SDR TRX for the STEMLab 122-16 ? 
I received my STEMLab 122-16 and i want build my SDR TRX on base of DiscoRedTRX.
Thanks in advance. Best Regards Walter / OE6WSF

pavel-demin, 2020-04-25:
I hadn't really planned it because I thought no one was interested in using it.

It is not very difficult to port this application to the 1-bit board.

If it's OK for you to build it from the source code and test it, then I could try adding the ported code to this repository. If you ask because you want to port the code yourself, it's even better.

joaquinbvw, 2020-06-03:
I also would want to implement the SDR transceiver on the STEMLab 122-16, can you tell me where to start? I imagine I would have to generate the bit file from vivado and upload it to the Red Pitaya right?

pavel-demin, 2020-06-03:
I'm afraid I can't tell you where to start without knowing exactly where you are now and in which direction you want to move. So, please provide more informations.

Have you looked at the applications listed at [this link](https://pavel-demin.github.io/stemlab-sdr-notes)? There is already one SDR transceiver and a pre-built SD card image that can be used without building anything.

This issue is about porting [this application](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-emb/) from the original Red Pitaya board to STEMlab 122.88-16 SDR. Do you want to do this porting? If so, then the first step is to edit the source code and make it compatible with STEMlab 122.88-16 SDR.