pavel-demin/stemlab-sdr-notes #19: Amature Band Plan on 160m and 80m Extended in Japan

Koichi-Koyama, 2020-04-28:
Dear Pavel-san,

This is Koichi Koyama (Mr) in Japan, and I have been enjoy using both STEMLab125-14 and STEMLab122-88-16 SDR with my homebrew amplifiers.  
In Japan, on 21st April, 2020, in amature bands on 160m and 80m, some additional frequencies  have been allocated as follows in addition to the existing frequencies. 

160m: 1,800KHz - 1,810kHz、and 1,825KHz - 1,875kHz
80m: 3,575KHz - 3,580kHz、and 3,662KHz - 3,680kHz

In addition to the extention of frequencies, SSB and SSTV modes can be used on 160m subject to application.  

I would be grateful if you could advise me if it is possible to modify the frequency settings of Japan region, or any plan to issue an update version of the software. 

At this moment, I can change the "Region" to "Extended" or other country in order to transmit within these frequencies. 

Your information would be greatly appreciated. 

Best regards,

Koichi Koyama
7N1ICA

pavel-demin, 2020-04-28:
I think that these changes should be done in the software (PowerSDR mRX PS) running on the PC. There is no information about regions in my code running on STEMlab 122.88-16 SDR.

pavel-demin, 2020-04-28:
I've just checked the code of PowerSDR mRX PS. The bands for Japan are defined on [lines 760-841 in database.cs](https://github.com/TAPR/OpenHPSDR-PowerSDR/blob/master/Project%20Files/Source/Console/database.cs#L760-L841). If I'm not mistaken, to change the bands, this file should be modified and the program should be rebuilt. So, I think that this question should be addressed to the developers of PowerSDR mRX PS.

Koichi-Koyama, 2020-04-29:
Thank you very much for your checking and information.  

Koichi Koyama
7N1ICA

差出人: Pavel Demin
送信日時: 2020年4月28日 21:27
宛先: pavel-demin/stemlab-sdr-notes
CC: Koichi-Koyama; Author
件名: Re: [pavel-demin/stemlab-sdr-notes] Amature Band Plan on 160m and 80m Extended in Japan (#19)

I've just checked the code of PowerSDR mRX PS. The bands for Japan are defined on lines 760-841 in database.cs. If I'm not mistaken, to change the bands, this file should be modified and the program should be rebuilt. So, I think that this question should be addressed to the developers of PowerSDR mRX PS.

closed 2020-06-28 by pavel-demin