pavel-demin/stemlab-sdr-notes #20: question: Is there a reason the "receiver compatible with HPSDR" does not work over the OTG port with the 122.88-16 board

satfan52, 2020-05-21:
Hello Pavel,

The Ethernet port of one of my modified boards is defective and I was not successful to repair it.  So  for that specific board  (clocked at 122.88 Mhz) I am using an external Ethernet Adapter over the OTG port.  This is how I noticed that the " transceiver compatible with  HPSDR works" quite well with an external Ethernet adapter with both the version of the software developed for the  125-14 or the 122.88-16 board. However, I also noticed today that the "Receiver Compatible with HPSDR" only works with an external adapter with the version of the software developed for 125-14 board.  With the  version of the software developed for 122.88-16 board,  the hpsdr receiver is simply not detected at all when an external adaptepr is used only the native Ethernet port works .. I have done the test with PowerSDR, Quisk and linhpsdr and your Alpine distro. Is there a reason for this ? Please note that just to be sure I verified those test results with two unmodified 125-14 and 122.88-16 boards.

Thanks
Regards
Peter

Thanks
Regards
Peter

pavel-demin, 2020-05-22:
Hello Peter,

The two sever processes in the receiver application are bound to the devices `eth0` and `mvl0`. I think that your problem can be solved by replacing `eth0` with the name of your new network interface in [start.sh](https://github.com/pavel-demin/stemlab-sdr-notes/blob/master/projects/sdr_receiver_hpsdr/app/start.sh).

Best regards,

Pavel

satfan52, 2020-05-22:
Thanks Pavel, yes replacing eth0 by eth1 in that file solves the problem for me

closed 2020-05-22 by satfan52

satfan52, 2020-06-29:
Hello Pavel,

Thanks for unifying the Alpine "SD card"  so the same can be used for  the 125-14 and 122.88-16 model. It is really handy for people who like me own the two type boards 

Unfortunately the work-around covered by this ticket does work any-longer as the structure of the start.sh file of the hpsdr transceiver for the 122.88-16 file has changed to be unified with the 125-14 one. The bottom line is that I can't change the eth0 (damaged on one of my boards) to the eth1 port (an external USB2 Ethernet adaper) manually in the start.sh file (as I used to do it before). Out of deduction, I found a work-around by simply ADDING the  line below in the start.sh file

ip link add mvl0 link eth1 address $address type macvlan mode passthru

....but the establishment of the network connection is slow; much slower than on my other healthy boards (even when using the OTG port) and also slower than with the work-around that I used before you unified the two sd_cards . It looks the system keeps trying to use eth0 and only when it gives up it tries eth1.

I am wondering if you could advise something else (smarter) to speed-up the establishment of the connection on the eth1 port on  that specific board ?.  

Thanks
Regards

PS: Please note that on the healthy boards, for both the 122.88-16 and the 125-14 boards, I can use either the OTG or Ethernet port without having to fiddle around with the start.sh file..

pavel-demin, 2020-06-29:
> Unfortunately the work-around covered by this ticket does work any-longer as the structure of the start.sh file of the hpsdr transceiver for the 122.88-16 file has changed to be unified with the 125-14 one.

I see two errors in this description of the problem:
 - This ticket isn't about transceiver. It's about receiver.
 - I have not changed the structure of the start.sh file

Are you sure that you're looking at the correct start.sh file? The path to the correct file is `apps/sdr_receiver_hpsdr_122_88/start.sh`.

satfan52, 2020-06-30:
Ok, the problem description is a bit confused, I agree, sorry about that, I will try to do better. 

In fact, it is the same problem with the receiver and hpsdr transceiver:  I have a board (122.88-16 like) where the Ethernet (eth0) is defective and I want to use the eth1 port instead. The problem is that with this specific unhealthy board,  when I connect the external adapter to the OTG port (eth1) and I connect nothing on the (eth0) Ethernet port, the conexion does NOT get established automatically over eth1 (as it is the case with an healthy board, regardless its type). 

So for the hpsdr transceiver app, which is in fact my main concern today,  I have added in the the corresponding start.sh file, the line cited in my previous ticket and this allows the connexion gets established via eth1 (ie, believe me it does not work without that line) ; I just wonders if there is a BETTER way to force the establishment of the eth1 conexion on that specific unhealthy board as I find it a bit slow.  

For some reasons I did NOT had that problem with the hpsdr transceiver app BEFORE you unified the Alpine distros for the two types boards, I had the problem only for hpsdr receiver (and you gave the work-around in this ticket). After the merge,  I have the problem for both the hpsdr receiver and the hpsdr transceiver so I ALSO need a work around for the hpsdr transceiver.  I found something myself but I am not sure it is optimal

DD8JM, 2020-06-30:
I installed the new software from Pavel.

I have made the following observation:

1. my bandpass filters are controlled by BCD code, when I switch on the Red Pitaya they rattle briefly, with the old version for about 1-2 seconds, because I did not use any pullup or pulldown resistors. With the new version it takes about 40 seconds until a stable state is reached at the BCD output of the Red Pitaya. 
2. the CPU load in % lower left in PowerSDR shows me with the new version about 5% less load, very pleasing.

**Everything else is fine, my thanks also to Pavel, I am very happy about the further development of the Red Pitaya software.** 

My guess is that the process boot or in other words the initialization of the Red Pitaya takes longer with the new version (35-40 seconds, old version only 1-2 seconds), so the BCD outputs are in an undefined state, which causes the following BCD/decimal decoder to switch undefined and my relays of the low pass filter start to rattle.

When the process is finished at the Red Pitaya the relays are stable and I can press the Start button without error message.

It's probably also due to my hardware constellation that this behavior of the software shows up in my case. Also in the old version the relays rattled briefly, but only for 1-2 seconds.
But before I change something at the hardware (pullup/pulldown resistors or a select release of the BCD/decimal decoder), I wait what the one or other has to report.

Many greetings
Joerg, DD8JM

pavel-demin, 2020-06-30:
> The problem is that with this specific unhealthy board, when I connect the external adapter to the OTG port (eth1) and I connect nothing on the (eth0) Ethernet port, the conexion does NOT get established automatically over eth1 (as it is the case with an healthy board, regardless its type).

Do I understand correctly that your healthy boards behave differently compared to the faulty board with exactly the same SD card image, exactly the same USB/Ethernet adapter and exactly the same network connection? If this is the case and a faulty board is required to reproduce the problem, I'm afraid that I cannot help you solve this problem.

What do you mean by "the connection does NOT get established automatically over eth1"? Do you mean that eth1 cannot obtain an IP address? Is eth1 connected to a router or directly to a PC?

satfan52, 2020-06-30:
Yes Pavel you understood correctly. Yes I meant no IP address is assigned by the DHCP server. The redpitaya is connected to a router and is on the same subnet than the PC

Well, I am afraid there is nothing to reproduce any-longer, the board just died under my eyes this morning and of course I have no idea why - something was obviously cooking, it just now refuses to boot, no boot messages sent anylonger over the usb console..... I will replace the XC7Z020 but I am not entirely convinced it is the hardware fault. Thanks anyway