pavel-demin/stemlab-sdr-notes #22: I installed the new software from Pavel, sdr_transceiver_hpsdr_122_88.

DD8JM, 2020-06-30:
I have made the following observation:

1.    my bandpass filters are controlled by BCD code, when I switch on the Red Pitaya they rattle briefly, with the old version for about 1-2 seconds, because I did not use any pullup or pulldown resistors. With the new version it takes about 40 seconds until a stable state is reached at the BCD output of the Red Pitaya.
2.    the CPU load in % lower left in PowerSDR shows me with the new version about 5% less load, very pleasing.

Everything else is fine, my thanks also to Pavel, I am very happy about the further development of the Red Pitaya software.

My guess is that the process boot or in other words the initialization of the Red Pitaya takes longer with the new version (35-40 seconds, old version only 1-2 seconds), so the BCD outputs are in an undefined state, which causes the following BCD/decimal decoder to switch undefined and my relays of the low pass filter start to rattle.

When the process is finished at the Red Pitaya the relays are stable and I can press the Start button without error message.

It's probably also due to my hardware constellation that this behavior of the software shows up in my case. Also in the old version the relays rattled briefly, but only for 1-2 seconds.
But before I change something at the hardware (pullup/pulldown resistors or a select release of the BCD/decimal decoder), I wait what the one or other has to report.

Many greetings
Joerg, DD8JM

pavel-demin, 2020-06-30:
Hello Joerg,

Thank you for testing the new version.

In previous versions, the FPGA was configured with the LED blinker configuration before booting Linux. In the new version, the FPGA remains unconfigured until an application is launched via the web interface or a start.sh script. This is why the GPIO pins remain in their default configuration for more than a few seconds.

Best regards,

Pavel

DD8JM, 2020-06-30:
Hello, Pavel,

Thanks for the quick answer, I understand.
Then I have to think about a hardware solution for my problem.
Finally, a question about the reduced CPU performance when using OpenHPSDR: Is it possible that using the new versions of Vitis 2020.1, Linux 5.4 Alpine and Linux 3.12 can lead to the improved performance I have observed?

I look forward to many more new software and features from you. ;)

Best regards,

Joerg, DD8JM

pavel-demin, 2020-06-30:
> Is it possible that using the new versions of Vitis 2020.1, Linux 5.4 and Alpine Linux 3.12 can lead to the improved performance I have observed?

I don't think there is anything in the new version that could cause a lower CPU load on your PC.

closed 2020-06-30 by pavel-demin

DD8JM, 2020-06-30:
I meant the CPU% display in OpenHPSDR or PowerSDR 3.4.9., I'm already aware that my PC has nothing to do with it, it's only about the traffic on the UDP channel between PC and Red Pitaya.
[image]